Octanis1/Octanis1-ROS #6: Implement closed loop controller for LIDAR motor

bitmorse, 2016-05-28:
LIDAR will not output data as long as RPM < 180. Power consumption of the spinning motor varies with temperature. Therefore a controller is required that keeps the motor spinning at 180RPM.

Controller software should be implemented on the Olimex A20. For hardware, an I2C PWM driver could be used which should be connected to the Olimex A20 I2C bus and the EPS supply. The motor requires 3V at ambient and 6V at -18C.

qcabrol, 2016-06-13:
ROS installed, xv11 driver not pulling data when the lidar is connected on uart0 (/dev/ttyS0), need to use a USB adapter or review the driver code.

bitmorse, 2016-06-13:
The lidar has the launchpad which you can use as a usb to uart adapter. The
adapter should appear as ttyACM or ttyAMA with the appropriate index (it
make 2 devices: 0 and 1).

bitmorse, 2016-07-02:
I2C motor controller set to arrive Monday and can be tested then. Use the I2C bus on UEXT1.

closed 2017-03-05 by bitmorse